mithi/hexapod #36: Rename params prop to be more descriptive for sliders and inputFields

mithi, 2020-06-11:
https://github.com/mithi/hexapod/blob/master/src/components/generic/Slider.js
https://github.com/mithi/hexapod/blob/master/src/components/generic/InputField.js

nitesh-kumar-sharma, 2020-06-17:
[https://github.com/mithi/hexapod/blob/master/src/components/generic/InputField.js](https://github.com/mithi/hexapod/blob/master/src/components/generic/InputField.js) 
showing not page not found

mithi, 2020-06-17:
@nitesh-kumar-sharma 

https://github.com/mithi/hexapod/blob/master/src/components/generic/NumberInputField.js

> https://github.com/mithi/hexapod/blob/master/src/components/generic/InputField.js
> showing not page not found

Thanks for reporting @nitesh-kumar-sharma 

Here's the updated link:

https://github.com/mithi/hexapod/blob/master/src/components/generic/NumberInputField.js

commit c6fd8573 (2020-06-18): refactor: Rename prop to be more descriptive (fixes #36) (#65)

Rename the param prop to attributes for Sliders and NumberInputFields for better clarity.
closed 2020-06-18 by mithi