mithi/hexapod #66: ValidityState is not working when passing any alphabet or character

nitesh-kumar-sharma, 2020-06-17:
As per the observation found in `NumberInputField` => `handleChange() `method propagating the exception when pass any character in place of number it directly goes to float conversion that means validation not handled properly.

[image]

mithi, 2020-06-18:
Thanks for submitting this issue, I'll fix it right away. 😅

mithi, 2020-06-18:
[image]

mithi, 2020-06-18:
[image]
[image]

commit a927b1e8 (2020-06-18): fix: Handle NaN parsed float (fixes #66) (#67)
closed 2020-06-18 by mithi