mithi/hexapod #117: Inverse Kinematics Page II - 2d Version

mithi, 2020-07-08:
It would be nice if we can control 2 or 3 parameters at the same time

by using a 2d slider, and 2 keys (up and down button) 
https://codepen.io/tyler-murphy/pen/tHsAu

# ideas

1. control rx and ry at the time
2. control tx, ty, tz at the same time

mithi, 2020-07-12:
# Page Idea 2

1. Be able to save poses and preselect saved named poses or saved ik configuration
in local storage: https://developer.mozilla.org/en-US/docs/Web/API/Window/localStorage

2. Be able to save named poses and ik configurations by pasting a json string in a text area and submitting it.
that we can deserialize and store it in local storage for quick access later. 
```js
{ type: "ikpose", name: "michael-jackson-thriller", ikParams: {rx, ry, rz, tx, ty, tz}, pose: {} }
{ type: "fkpose", name: "elvis-stay-alive", pose: {} }
```
We should also perform sanity checks 

3.  We can also perform animations given a json object specification for example. 
```js
{ 
    type: "movement",
    name: "stay-alive-dance", 
    poseNames: ["point-up", "point-down", "look-left", "look-right"],
    delay: [5000, 2000, 250, 1200]
 }
```

4. We can then pull up these named movements, and show the animation in a loop.

EllingtonK, 2020-09-19:
Hi mithi.  I am really intrigued by your Inverse Kinematics Page Joystick idea.  I would like to take up this issue also if that's okay with you.

mithi, 2020-09-20:
> Hi mithi. I am really intrigued by your Inverse Kinematics Page Joystick idea. I would like to take up this issue also if that's okay with you.

Hey, this is so cool. I opened up a new issue solely for this. https://github.com/mithi/hexapod/issues/141 We can continue the conversation there, if that's okay with you.

mithi, 2020-09-20:
@EllingtonK 

You can respond directly to issue https://github.com/mithi/hexapod/issues/141 for any clarifications!
You can also checkout the [contribution guidelines](https://github.com/mithi/hexapod/blob/master/CONTRIBUTING.md) and [commit styleguide!](https://github.com/mithi/hexapod/wiki/A-Commit-Style-Guide)