mithi/hexapod #134: Fix lighthouse compaints

mithi, 2020-07-26:
Open dev tools and generate report with lighthouse 
- Use video instead of gif
- Use passive scroll listeners

[image]

[image]

mithi, 2020-07-30:
[image]

thesanjeevsharma, 2020-08-12:
Hi @mithi, 
I would like to take up this issue.

mithi, 2020-08-13:
Sure! Go ahead! Enjoy! :D

vparine, 2020-10-01:
@mithi Can I work on this?

mithi, 2020-10-01:
> @mithi Can I work on this?

Sure! Thank you for your interest! 
Don't forget to read the [contributing guidelines](https://github.com/mithi/hexapod/blob/master/CONTRIBUTING.md) and the recommended [commit style guide](https://github.com/mithi/hexapod/wiki/A-Commit-Style-Guide)! Thank you!

And let me know if you have any questions!

mithi, 2021-11-01:
> Hi @mithi, can I pick this up?

Hi @disawalva ,
Sorry for the late reply! Sure :) Please let me know if you have any questions. Thanks!

disawalva, 2021-11-01:
Sure :)
Thank you!