mithi/hexapod #138: Move plotter from `src/hexapod` to `src/templates`

mithi, 2020-08-07:
Move this whole file to `src/templates` directory. Update all dependents and dependencies, make sure nothing is broken
https://github.com/mithi/hexapod/blob/cdd2f1e89cf3b2a1bffb32c4769d2f6a917e8b35/src/hexapod/plotter.js

Rational: the hexapod subdirectory contains all the files to compute the orientations and points of the hexapod, 
it shoudn't be in charge of displaying the hexapod. Also we want to move a couple of steps forward to making the `hexapod` subdirectory it's own package / library which can be called by any application. 

https://github.com/mithi/hexapod/blob/cdd2f1e89cf3b2a1bffb32c4769d2f6a917e8b35/src/hexapod/plotter.js#L1-L100

lashewi, 2020-08-09:
Can I work on this?

mithi, 2020-08-09:
> Can I work on this?

Sure! Make sure nothing is broken!

2Shar18, 2020-08-09:
> Can I work on this?

Hey, have you started working on this? 
If not can I take this issue?

mithi, 2020-08-09:
> > Can I work on this?
> 
> Hey, have you started working on this?
> If not can I take this issue?

Sure! Make sure you don't break anything! :D

PR #140 (2020-08-09): Move: hexapod/plotter to templates/plotter
Closes #138 
This is my first contribution and PR, hope I am doing it in a proper way.
After making the changes, I tried testing and running the project with `npm test` and `npm start` with complete success.
closed 2020-08-09 by mithi