mithi/hexapod #139: Remove unused code, make sure it doesn't break anything

mithi, 2020-08-08:
Find and use code that are unnecessary

https://github.com/mithi/hexapod/blob/cdd2f1e89cf3b2a1bffb32c4769d2f6a917e8b35/src/templates/plotParams.js#L280-L287

https://github.com/mithi/hexapod/blob/cdd2f1e89cf3b2a1bffb32c4769d2f6a917e8b35/src/templates/plotParams.js#L9-L10

https://github.com/mithi/hexapod/blob/cdd2f1e89cf3b2a1bffb32c4769d2f6a917e8b35/src/hexapod/VirtualHexapod.js#L302-L307

EllingtonK, 2020-09-19:
Hi mithi.  I am a first timer.  I would like to take up this issue.

mithi, 2020-09-20:
> Hi mithi. I am a first timer. I would like to take up this issue.

Sure. Make sure your change doesn't break anything!

mithi, 2020-09-20:
> Hi mithi. I am a first timer. I would like to take up this issue.

Sure, make sure it doesn't break anything!
You can also checkout the [contribution guidelines](https://github.com/mithi/hexapod/blob/master/CONTRIBUTING.md) and [commit styleguide!](https://github.com/mithi/hexapod/wiki/A-Commit-Style-Guide)

EllingtonK, 2020-10-01:
Hi mithi.  I was going to work on this issue first.  I thought I would be able to download the zip into Apache Netbeans and run it, but it doesn't seem to run.  So I wanted to create a separate branch to work on editing the code, but it doesn't look like I'm allowed to add a branch.  Any suggestions?

mithi, 2020-10-01:
> Hi mithi. I was going to work on this issue first. I thought I would be able to download the zip into Apache Netbeans and run it, but it doesn't seem to run. So I wanted to create a separate branch to work on editing the code, but it doesn't look like I'm allowed to add a branch. Any suggestions?

I don't know how apache netbeans work but you can work on it on your local machine or https://codesandbox.io/.

You can do follow the following steps
1. Fork the mithi/hexapod repository to your own account using the fork button on the top-right of the GitHub page.
2. Clone your repository and cd into it: `git clone https://github.com/username/hexapod.git`
3. Inside the directory, add upstream, ie: `git remote add upstream https://github.com/mithi/hexapod.git`
4. Create and checkout the branch you'd like to work on
```
$ git checkout master
$ git checkout -b your-branch-name
```
Commit and push as you like to your own fork
```
git push origin your-branch-name
```
See also: https://github.com/mithi/hexapod/wiki/Simple-Git-Workflow-guide

EllingtonK, 2020-10-02:
OK, thank you.  I have it cloned correctly now.

EllingtonK, 2020-10-11:
Hi mithi. I've modified the plotParams.js file and tried to run the Test Suite, but I'm not sure it's working.  The screen flashes for a few minutes, but nothing ever displays in the Test summary window.  I thought it was supposed to display a success or failure message for each test.   Am I missing something?

[image]

Also, I saw that there was a new commit to your branch, but when I try to fetch the changes and rebase, I get an error message.

[image]

EllingtonK, 2020-10-17:
OK, the test suite issue appears to be a problem with my browser, not codesandbox.  I'm still not sure why I can't rebase though.

mithi, 2020-10-17:
> OK, the test suite issue appears to be a problem with my browser, not codesandbox. I'm still not sure why I can't rebase though.

Hey, did you try

```
git remote add upstream https://github.com/mithi/hexapod.git
git fetch upstream
git checkout master
git rebase upstream/master
```

Interesting related readings
- https://gist.github.com/ravibhure/a7e0918ff4937c9ea1c456698dcd58aa
- https://blog.algolia.com/master-git-rebase/
- https://www.brad.net.nz/blog/2017/07/merging-a-git-repository-from-upstream-when-rebase-wont-work/

mithi, 2020-10-17:
@EllingtonK 

After doing `git fetch upstream` Did you also try checking out your branch and doing `get rebase master` ?

EllingtonK, 2020-10-18:
Yes, but something's still wrong.  It tells me the rebase is successful, but this is what I show when I try a git status.

[image]

And on my github page I still show I'm behind.

[image]

mithi, 2020-10-18:
@EllingtonK 

do `git push --force`

mithi, 2020-10-18:
> git push --force

@EllingtonK 

I suggest you do the exercises here: https://onlywei.github.io/explain-git-with-d3/
So that you get more familiar with how git works visually.

I also recommend you install the vscode plugin `git graph`

https://ardalis.com/git-graph-visualizes-branches-in-vs-code-for-free/

EllingtonK, 2020-10-18:
OK thanks.

mithi, 2020-10-19:
@EllingtonK Is is also good: https://learngitbranching.js.org/

closed 2020-10-19 by mithi